itaybre/CameraController #3: Not working with Logitech C615

codetheweb, 2020-08-10:
I have a Logitech C615 webcam. If it's plugged in before starting the app, the app launches but never displays the settings window. If I launch the app before plugging it in, it shows the settings window but freezes up as soon as I plug it in.

Thanks for all the time you've put into this, I really appreciate it as I haven't been able to find a good macOS camera settings app.

Itaybre, 2020-08-10:
Thanks for the report, may I know if you tried with the latest version 1.1 or the previous one 1.0?

Does the app crash after freezing?

codetheweb, 2020-08-10:
It's 1.1.

It does not crash after freezing, but does pin the CPU at 100% until I force it to quit.

Itaybre, 2020-08-11:
Probably got stuck [here](https://github.com/Itaybre/CameraController/blob/master/CameraController/UVC/Extensions/IOUSBConfigurationDescriptorPtr%2BUVC.swift#L25)

Can I ask you to run the app from the source code and check where it got stuck?
The output of `ioreg -p IOUSB -w0 -l` will also help

codetheweb, 2020-08-11:
Here's the output when starting with the camera plugged in:

```
2020-08-10 20:08:21.537344-0500 CameraController[30939:628745] Metal API Validation Enabled
2020-08-10 20:08:25.567910-0500 CameraController[30939:628745] [plugin] AddInstanceForFactory: No factory registered for id <CFUUID 0x600000275e40> F8BB1C28-BAE8-11D6-9C31-00039315CD46
2020-08-10 20:08:25.632296-0500 CameraController[30939:628745]  HALC_ShellDriverPlugIn::Open: Can't get a pointer to the Open routine
2020-08-10 20:08:25.657369-0500 CameraController[30939:628745] [plugin] AddInstanceForFactory: No factory registered for id <CFUUID 0x600000211f60> 30010C1C-93BF-11D8-8B5B-000A95AF9C6A
```

Output of `ioreg -p IOUSB -w0 -l` (I went ahead and removed everything that didn't seem relevant):

```
+-o Root  <class IORegistryEntry, id 0x100000100, retain 23>
  | {
...
  |   | +-o HD Webcam C615@02231000  <class AppleUSBDevice, id 0x10000282b, registered, matched, active, busy 0 (3 ms), retain 18>
  |   |     {
  |   |       "sessionID" = 31902761108411
  |   |       "idProduct" = 2092
  |   |       "iManufacturer" = 0
  |   |       "bDeviceClass" = 239
  |   |       "bMaxPacketSize0" = 64
  |   |       "bcdDevice" = 17
  |   |       "iProduct" = 2
  |   |       "iSerialNumber" = 1
  |   |       "bNumConfigurations" = 1
  |   |       "Bus Power Available" = 250
  |   |       "USB Address" = 19
  |   |       "Built-In" = No
  |   |       "locationID" = 35852288
  |   |       "bDeviceSubClass" = 2
  |   |       "bcdUSB" = 512
  |   |       "USB Product Name" = "HD Webcam C615"
  |   |       "PortNum" = 1
  |   |       "non-removable" = "no"
  |   |       "kUSBSerialNumberString" = "E0500BC0"
  |   |       "bDeviceProtocol" = 1
  |   |       "AppleUSBAlternateServiceRegistryID" = 4294977577
  |   |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   |       "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
  |   |       "Device Speed" = 2
  |   |       "kUSBCurrentConfiguration" = 1
  |   |       "idVendor" = 1133
  |   |       "IOGeneralInterest" = "IOCommand is not serializable"
  |   |       "kUSBProductString" = "HD Webcam C615"
  |   |       "USB Serial Number" = "E0500BC0"
  |   |       "IOClassNameOverride" = "IOUSBDevice"
  |   |     }
  |   |     
...
```

Also recorded a trace in Instruments of the CPU being pinned at 100%:

[Instruments.trace.zip](https://github.com/Itaybre/CameraController/files/5054234/Instruments.trace.zip)

Hope that helps!

Itaybre, 2020-08-11:
Just found a tool to get good USB Description [USB Prober.zip](https://github.com/Itaybre/CameraController/files/5054537/USB.Prober.zip) (It is not signed but It was made by Apple at OS X 10.9 era)

Please send me the output you get from there, this should be the last log I need, because I will be able to manually run the code with this.

You can see the save output inside Bus Probe -> Top Right
[image]

codetheweb, 2020-08-11:
[USB Bus Probe.txt](https://github.com/Itaybre/CameraController/files/5057488/USB.Bus.Probe.txt)

Itaybre, 2020-08-11:
I think I found the reason and why it didn't happen on my machine. I wasn't advancing the iterator when skipping non relevant interfaces, thus creating a loop. And this never happened on my cameras because the first interface was the correct one for all my devices.

Can you try this version? [CameraController.zip](https://github.com/Itaybre/CameraController/files/5057726/CameraController.zip)

codetheweb, 2020-08-11:
It doesn't crash/freeze now. 👍 
I can see the camera preview, but cannot change any of the settings.

[image]

Itaybre, 2020-08-12:
Reading the USB Standard again, it seems I was missing the Interface ID on the requests to the device.
Because it was 0 for me, it never failed, but in your case it is 2
[CameraController.zip](https://github.com/Itaybre/CameraController/files/5060166/CameraController.zip)

Itaybre, 2020-08-12:
> Reading the USB Standard again, it seems I was missing the Interface ID on the requests to the device.
> Because it was 0 for me, it never failed, but in your case it is 2
> [CameraController.zip](https://github.com/Itaybre/CameraController/files/5060166/CameraController.zip)

@codetheweb Forgot to mention, can you try the attached build? 
I hope it works, if not, I will have to get a similar camera

codetheweb, 2020-08-12:
It works great now, thank you so much!

closed 2020-08-12 by codetheweb