itaybre/CameraController #11: Dell UltraSharp Webcam (WB7022) Causes app to quit on launch

scottio200, 2021-07-08:
Please enter the following data:

Brand: Dell
Camera: WB7022
Have you tried many USB ports? Yes

App quits on launch with the camera connected, if I unplug camera app loads, as soon as I connect it again the app will quit.

Itaybre, 2021-07-08:
Thanks for the report @scottio200, do you have the crash log?

scottio200, 2021-07-14:
> Thanks for the report @scottio200, do you have the crash log?

`Process:               CameraController [90403]
Path:                  /Volumes/VOLUME/*/CameraController.app/Contents/MacOS/CameraController
Identifier:            com.itaysoft.CameraController
Version:               1.2 (3)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           CameraController [90403]
User ID:               501

Date/Time:             2021-07-14 09:32:58.931 +0200
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Bridge OS Version:     5.4 (18P4663)
Anonymous UUID:        ACCE28B4-A417-40E0-3CB0-0346A5D764B2

Sleep/Wake UUID:       3BF61DDF-41BA-4BC7-BDCC-941A24C66EA5

Time Awake Since Boot: 310000 seconds
Time Since Wake:       2900 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [90403]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.itaysoft.CameraController 	0x0000000101ea8111 UnsafeMutablePointer<A>.browseDescriptor(_:_:_:_:_:) + 5873
1   com.itaysoft.CameraController 	0x0000000101ea69fc UnsafeMutablePointer<A>.proccessDescriptor() + 556
2   com.itaysoft.CameraController 	0x0000000101e4abf6 AVCaptureDevice.usbDevice() + 2518
3   com.itaysoft.CameraController 	0x0000000101e69611 UVCDevice.init(device:) + 81
4   com.itaysoft.CameraController 	0x0000000101e69574 UVCDevice.__allocating_init(device:) + 52
5   com.itaysoft.CameraController 	0x0000000101e23dd4 CaptureDevice.init(avDevice:) + 228
6   com.itaysoft.CameraController 	0x0000000101e23cd9 CaptureDevice.__allocating_init(avDevice:) + 41
7   com.itaysoft.CameraController 	0x0000000101e269fd closure #1 in DevicesManager.().init() + 93
8   com.itaysoft.CameraController 	0x0000000101e26a26 thunk for @callee_guaranteed (@guaranteed AVCaptureDevice) -> (@owned CaptureDevice, @error @owned Error) + 22
9   com.itaysoft.CameraController 	0x0000000101e26a84 partial apply for thunk for @callee_guaranteed (@guaranteed AVCaptureDevice) -> (@owned CaptureDevice, @error @owned Error) + 20
10  libswiftCore.dylib            	0x00007fff2c8f82bc Collection.map<A>(_:) + 524
11  com.itaysoft.CameraController 	0x0000000101e26492 DevicesManager.().init() + 978
12  com.itaysoft.CameraController 	0x0000000101e24a64 DevicesManager.__allocating_init() + 36
13  com.itaysoft.CameraController 	0x0000000101e24a2c one-time initialization function for shared + 28
14  libdispatch.dylib             	0x00007fff2038f806 _dispatch_client_callout + 8
15  libdispatch.dylib             	0x00007fff2039098c _dispatch_once_callout + 20
16  libswiftCore.dylib            	0x00007fff2cc2f16a swift_once + 26
17  com.itaysoft.CameraController 	0x0000000101e24aa4 DevicesManager.shared.unsafeMutableAddressor + 52
18  com.itaysoft.CameraController 	0x0000000101e23126 ContentView.init() + 54
19  com.itaysoft.CameraController 	0x0000000101e31980 AppDelegate.applicationDidFinishLaunching(_:) + 160
20  com.itaysoft.CameraController 	0x0000000101e32642 @objc AppDelegate.applicationDidFinishLaunching(_:) + 146
21  com.apple.CoreFoundation      	0x00007fff2062810d __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
22  com.apple.CoreFoundation      	0x00007fff206c3b6c ___CFXRegistrationPost_block_invoke + 49
23  com.apple.CoreFoundation      	0x00007fff206c3aea _CFXRegistrationPost + 454
24  com.apple.CoreFoundation      	0x00007fff205f938e _CFXNotificationPost + 736
25  com.apple.Foundation          	0x00007fff21367dc8 -[NSNotificationCenter postNotificationName:object:userInfo:] + 59
26  com.apple.AppKit              	0x00007fff22e3a9c0 -[NSApplication _postDidFinishNotification] + 305
27  com.apple.AppKit              	0x00007fff22e3a712 -[NSApplication _sendFinishLaunchingNotification] + 208
28  com.apple.AppKit              	0x00007fff22e378b1 -[NSApplication(NSAppleEventHandling) _handleAEOpenEvent:] + 541
29  com.apple.AppKit              	0x00007fff22e37507 -[NSApplication(NSAppleEventHandling) _handleCoreEvent:withReplyEvent:] + 665
30  com.apple.Foundation          	0x00007fff21393516 -[NSAppleEventManager dispatchRawAppleEvent:withRawReply:handlerRefCon:] + 308
31  com.apple.Foundation          	0x00007fff21393386 _NSAppleEventManagerGenericHandler + 80
32  com.apple.AE                  	0x00007fff264067f3 0x7fff263fa000 + 51187
[... 537 lines cut ...]
USB Device: Dell Webcam WB7022
USB Device: 4-Port USB 3.0 Hub
USB Device: USB2.1 Hub
USB Device: Magic Keyboard
USB Device: USB2.1 Hub
USB Device: USB2.0 Hub
USB Device: Elgato Wave:3
USB Device: Stream Deck
USB Device: USB2.1 Hub
USB Device: HID Device
USB Device: Dell Presence Sensor WB7022
USB Device: 4-Port USB 2.0 Hub
USB Device: USB Audio
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 85.0
Thunderbolt Bus: MacBook Pro, Apple Inc., 85.0
Thunderbolt Device: Thunderbolt 3 Dock Plus, Belkin, 3, 45.1
`

Itaybre, 2021-07-14:
@scottio200 as I thought, it crashing while browsing the Camera Information.
Can you share me full USB descriptor of your machine? Seems like this camera descriptor is different from the ones I tested.
You can use the attached app (it is not signed because I extracted it using this guide: https://superuser.com/a/814339)
This will help me find the problem

[image]
[USB Prober.zip](https://github.com/Itaybre/CameraController/files/6817450/USB.Prober.zip)

scottio200, 2021-07-15:
[USB Bus Probe.txt.zip](https://github.com/Itaybre/CameraController/files/6824242/USB.Bus.Probe.txt.zip)

USB Probe attached

Itaybre, 2021-07-15:
I think I found the problem, can you try this build?
[CameraController.zip](https://github.com/Itaybre/CameraController/files/6825344/CameraController.zip)

scottio200, 2021-07-16:
That is working, thanks a lot!

closed 2021-07-16 by scottio200