itaybre/CameraController #12: Support Apple Silicon (M1 cpu)?

ThomasWaldmann, 2021-07-19:
Thanks for providing this app - found it while searching for a free app that just shows the picture of a camera.

As it is, it somehow works (shows the camera live picture) on a Macbook Air 2020 (M1), but tends to be a bit slow as it is translated for the M1 first. Guess this could be fixed by recompiling.

I tested it with:
- builtin Apple facetime camera
- Teslong USB microscope

Notable: all the settings sliders in the app are inactive - is this due to my camera hw, due to M1?

Itaybre, 2021-07-19:
@ThomasWaldmann I will compile it for M1 in the next version.

And regarding to all the sliders being disabled, it is due to special permissions required to interact with Apple hardware, it should work fine with USB cameras (happens with all Macs with T1, T2 and M1 chips)

ThomasWaldmann, 2021-07-19:
Great, thanks!

About the sliders: they did not work with my USB camera either.

Itaybre, 2021-08-23:
Just release v1.3: https://github.com/Itaybre/CameraController/releases/tag/1.3
It adds Apple Silicon arch

closed 2021-08-23 by Itaybre