itaybre/CameraController #13: Big Sur 11.5 unable to launch app after download

UziKhana, 2021-08-09:
Expected behaviour:
1. download app
2. launch app
3. get third party app warning/caution note
4. goto Settings > Security & Privacy and click Open Anyway
5. app opens

What is happening:

1. download app
2. launch app
3. get third party app warning/caution note
4. get CameraController quit unexpectedly message (see screenshot) 
[image]

4. goto Settings > Security & Privacy, there is no button for Open Anyway
5. click Reopen, settings > Security & Privacy flickers a bit but still o button for Open Anyway

Other symptom:
1. app icon disabled (screenshot attached) 
[image]

Result:
Unable to open the app since first download

Itaybre, 2021-08-11:
@UziKhana thanks for the ticket, may I know which camera are you using?

Can you also try this build: https://github.com/Itaybre/CameraController/files/6825344/CameraController.zip ?
This one has a fix for some cameras

Itaybre, 2021-08-23:
Should be fixed on [1.3](https://github.com/Itaybre/CameraController/releases/tag/1.3)
Please reopen if not fixed

closed 2021-08-23 by Itaybre

UziKhana, 2021-08-23:
Hey thanks for taking this on. I shall try shortly. I am using Dell 4k ultrasharp camera (sorry I missed responding to your earlier message).

UziKhana, 2021-08-23:
Still getting this, I have a also tried CamLink 4k

[image]