itaybre/CameraController #14: App does not save profiles after Auit

maishsk, 2021-08-12:
**Describe the bug**[image]When I close the app, the profiles that I saved are deleted. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open Camera Controller
2. Make some changes to the settings
3. Save profile - "My profile"
4. Apply "My profile" -> Settings apply successfully
5. Apply "Camera Default" -> Settings apply successfully
6. Apply "My profile" -> Settings apply successfully
7. Quit Camera Controller
8. Open Camera controller
9. The only profile available is "Camera Default" 

**Expected behavior**
I would expect to have the profile I created still there in the app so I Can apply the settings I configured 

**Screenshots**
[image]

[image]

After restart 
[image]

**Desktop (please complete the following information):**
 - Catalina 10.15.7
 - Lifecam Cinema

**Additional context**
Add any other context about the problem here.

Itaybre, 2021-08-23:
I could not reproduce this bug, but I will keep trying

maishsk, 2021-08-24:
Update from my side, I restarted the Mac, and now it saves settings, could be one of those weird edge cases