itaybre/CameraController #15: macOS Monterey all settings greyed out

affrae, 2021-08-13:
**Describe the bug**

in macOS Monterey all settings greyed out

**To Reproduce**
Steps to reproduce the behavior:
1. Launch app
2. Choose your camera

**Expected behavior**
Able to set settings for the camera

**Screenshots**

**_Note: camera does show up I have just blacked out the camera image._**

[image]
[image]
[image]

**Desktop (please complete the following information):**
 - Version: 1.2.1(4) from https://github.com/Itaybre/CameraController/files/6825344/CameraController.zip
 - Camera: Logitech Brio
 - macOS: 12.0 Monterey Public Beta 5

**Additional context**
Hi! I know it is a beta OS but opening here for reference, not expecting an immediate fix :)

Itaybre, 2021-08-20:
@affrae I would like to see the USB Descriptor for this camera.
Can you share it using [this](https://github.com/Itaybre/CameraController/issues/11#issuecomment-880016985) method?

affrae, 2021-08-22:
@Itaybre Here you go - hope this helps!

[USB Bus Probe.txt](https://github.com/Itaybre/CameraController/files/7028315/USB.Bus.Probe.txt)

Itaybre, 2021-08-24:
The descriptor seems to be fine, I will be testing this on a VM this weekend and already found a provider for the webcam locally in case the issue is with the camera and not Monterey

affrae, 2021-08-24:
Wow - thank you!!

enquora, 2021-08-24:
> The descriptor seems to be fine, I will be testing this on a VM this weekend and already found a provider for the webcam locally in case the issue is with the camera and not Monterey

The problem occurs with a Microsoft Lifecam also - it’s Monterey rather than any specific camera.

tfpbatelaan, 2021-08-25:
Hey there, this also occurs on Big Sur on my 15-inch mid-2015 Macbook Pro. It doesn't even work with the built-in!

Itaybre, 2021-08-25:
> Hey there, this also occurs on Big Sur on my 15-inch mid-2015 Macbook Pro. It doesn't even work with the built-in! 

That's expected behavior with built-in camera, apple doesn't allow low level communication with it

bkboy21, 2021-08-31:
hey , im using a mac mini and this webcam : 
https://www.amazon.com/dp/B08DXSG5QR/ref=cm_sw_r_apan_glt_fabc_XKFH6SSFZXRYQSP0XB7P?_encoding=UTF8&psc=1

all settings work but pan\zoom\tilt wont work . anything we can do about that ?
thank you for your contribution !!!!

affrae, 2021-08-31:
I should point out that LogiTune is not working either, so we may need to be patient here as it could be an underlying Beta issue, or require some rework generally.

apobst83, 2021-09-02:
+1 on the greyed out sliders in monterey. I am using a Dell Ultrasharp 4k Webcam.

jrykner, 2021-09-13:
+1 with latest Mac OS12 Beta 6, greyed out sliders, with logitech c922 webcam, it does recognize and preview, on Mac M1 but all options are greyed out

mattiasstahre, 2021-09-22:
+1 All settings are greyed out. I'm on Beta 7

don-sch, 2021-10-05:
+1 Any updates on this? I'm on beta 12. No worries if you're still figuring this one out.

Itaybre, 2021-10-31:
Fix merged, working on another fix and will upload a new build

closed 2021-10-31 by Itaybre

affrae, 2021-10-31:
Thank you, sir! Without me pitching the feature, you should look into sponsorship 😈

olsw, 2021-11-01:
Latest build failed so can't get this version, could you take a look please

affrae, 2021-11-01:
Confirmed [1.3.1 Release](https://github.com/Itaybre/CameraController/releases/tag/1.3.1) binary working for me on Monterey.

Itaybre, 2021-11-01:
@olsw you can download build 1.3.1 from the release now

> Thank you, sir! Without me pitching the feature, you should look into sponsorship 😈

Unfortunately Github Sponsors is not available on my country and I did setup a Ko-fi site, but still haven't been able to give it some love.
I may add the sponsor link once I finish v2.0 with menu bar configuration

affrae, 2021-11-01:
> Unfortunately Github Sponsors is not available on my country 

Oh manh I forgot about that, sorry.

4ndv, 2022-01-03:
Sadly, but for me it's still the same. Tested on 12.0.1, then updated to 12.1 but all the settings is still greyed out :(

affrae, 2022-01-03:
@4ndv what camera?

bobbynj81, 2022-01-10:
@affrae Everything is grayed out for me as well on Mac Monterey, M1 Mini with a Logitech Brio.

affrae, 2022-01-10:
Hi folks I'm just another user like y'all. I opened the issue here and since then the issue has been resolved for me. I can't help much here. 

@Itaybre do yo want folks posting on this issue, or close this one off and get folks to open a new one?

rinogo, 2023-03-30:
Same issue, this time with an "HD Webcam eMeet C980 Pro".

vogithub1, 2024-02-29:
Same issue - Macbook Air 13 2015 FaceTime HD Camera Monterey 12 3 1

SoftwareApe, 2025-05-06:
Same issue with OBSBot Meet 2.