itaybre/CameraController #20: Add a way to reset values to the default

globau, 2021-09-01:
After playing around with settings there's no easy way to reset back to their default values.

For the image settings (brightness, contrast, etc) that isn't too bad you can approximately set them to a 50% value (see also #19).
However, for some other controls the default value isn't obvious.

Itaybre, 2021-09-01:
You can use the Camera Default preset to restore all values, but not for a single one.

Seems like a good enhancement, added to 1.4

markbenliyan, 2022-01-19:
seconding this. Was it added? I don't see it yet

Itaybre, 2022-01-19:
@markbenliyan webcam setting generally reset when you disconnect it
Have you tried doing that?

markbenliyan, 2022-02-23:
> @markbenliyan webcam setting generally reset when you disconnect it
> Have you tried doing that?

Yes, I tried doing that and still stuck w the changes 1 month later. Honestly really disappointed not sure how to reset to default.

wookayin, 2022-04-16:
The stalebot is so great, It helps kills issues and feature requests.

Itaybre, 2022-04-16:
> The stalebot is so great, It helps kill issues and feature requests.

This is an open source project, anyone can make suggestions, requests features and PRs.
Unfortunately I don't have the bandwidth to work on every enhancement or issue.

I will add `enhancement` tag to the exempted list so they dont get closed in the future