itaybre/CameraController #22: Brightness can only be increased, not decreased

benblb, 2021-09-30:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to basic settings
2. Click on brightness
3. Drag slider all the way to the left
4. See "normal" brightness. 

**Expected behavior**
Ability to decrease brightness in addition to increasing. Image should get darker than the cam's default. 

**Desktop (please complete the following information):**
 - Big Sur 11.6
 - Camera Depstech DW49 (4k)

Itaybre, 2021-10-31:
@benblb I have not been able to reproduce this, can you submit a demo of the issue?

benblb, 2021-11-01:
Absolutely! https://imgur.com/a/E40KEVm

When the slider is all the way to the left, that is what the camera looks like without any modifications (i.e. default brightness).

replete, 2021-11-24:
I am getting this with bug also with an Aukey 1080p HD webcam, which shows up as 'PC-LM1E'.

I increased the brightness via the slider, but moving the slider to the left results in no changes being applied - and there appears to be no way to reset this.

closed 2022-01-26 by stale[bot]