itaybre/CameraController #23: CLI mode?

geftactics, 2021-10-04:
Great to see an app under active development that's providing this functionality.

I have a Logitech StreamCam, and am in the UK - so 50Hz power, but the camera defaults to 60Hz each time it's reconnected, so I end up with lines all over my face. Using this app, I can set the camera to 50Hz, but I have to manually do this all of the time.

Is it possible to control the app via command line? For example, passing the parameter you want to set and a value as a command line argument... or maybe even just recalling a profile that had previously been built the the UI? This would allow me to have a LaunchAgent that could call it at the right time! - Also open to other suggestions, maybe the functionality I need is already there via another means?

Many thanks!

Itaybre, 2021-10-05:
It is not a bad idea, I have planned to add an option to apply a saved profile when connecting a camera, but didn't had time to implement it

mcgregor94086, 2021-11-25:
I am also looking for a CLI or API version that I can use to programmatically set parameters.  See my post here: https://apple.stackexchange.com/questions/431339/how-to-set-web-camera-properties-on-macos?noredirect=1#comment617894_431339.

Here is how I programmatically set these on Linux, but I don't know how to do the equivalent on MacOS.

#!/bin/bash
camera_id=$1
full_image_file_name=$2

v4l2-ctl -d $camera_id -c brightness=64
v4l2-ctl -d $camera_id -c contrast=100
v4l2-ctl -d $camera_id -c saturation=66
v4l2-ctl -d $camera_id -c hue=0
v4l2-ctl -d $camera_id -c white_balance_temperature_auto=0
v4l2-ctl -d $camera_id -c gamma=500
v4l2-ctl -d $camera_id -c white_balance_temperature=4600
v4l2-ctl -d $camera_id -c sharpness=66
v4l2-ctl -d $camera_id -c backlight_compensation=1
v4l2-ctl -d $camera_id -c exposure_auto=1
v4l2-ctl -d $camera_id -c exposure_absolute=233
v4l2-ctl -d $camera_id -c exposure_auto_priority=0

v4l2-ctl -d $camera_id --stream-mmap --stream-count=1 --stream-to=$full_image_file_name &

wookayin, 2021-11-25:
FYI: https://github.com/makenai/node-uvc-control is another alternative (works on Mac as well)

geftactics, 2021-11-26:
oh nice!! After installing `uvcc` via `nvm`, I can now just do...
`uvcc set power_line_frequency 1`

closed 2022-01-26 by stale[bot]

dimitre, 2025-07-05:
I've been using a similar project to achieve this same result
https://github.com/jtfrey/uvc-util
```
../../../uvc-util -I 0 -s power-line-frequency=2 -s auto-exposure-mode=1 -s exposure-time-abs=76 -s gain=0 -s brightness=15 -s contrast=36 -s backlight-compensation=10 -s gamma=159 
```