itaybre/CameraController #28: Not working with Zoom

zain, 2021-11-16:
**Describe the bug**
All settings get reset to default when Zoom tries to use the camera.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a profile in CameraController
2. Join a zoom call
3. Notice the settings you just set are gone. 

**Desktop (please complete the following information):**
 - Version: Latest
 - Camera: Razer Kiyo

liorgrossman, 2022-01-12:
Same here, opening a Zoom call resets all settings.
I think it has to do with macOS Monterey. What version of macOS do you use, @zain ?

Related discussions:

- https://community.zoom.com/t5/Chat/Anybody-else-having-problem-with-Logitech-webcam-with-the-new/m-p/23460
- https://forums.macrumors.com/threads/macos-monterey-broke-my-usb-webcam.2320189/page-2

Itaybre, 2022-01-19:
Enabling "Push Settings" inside Preferences, should allow you to keep using your setting inside Zoom.

You might need to disable "Read Settings" though

closed 2022-03-13 by stale[bot]