itaybre/CameraController #29: App crashes upon camera connect

KindaNeatdoe, 2021-11-25:
**Describe the bug**
App closes when my webcam connects. I have a usb-c hub which doesn't really like my camera and it just disconnects randomly for a second and comes up again. 

**To Reproduce**
Steps to reproduce the behavior:
1.  Open application
2. Connect another camera
3. Wait for camera to get detected
4. App crashes

**Expected behavior**
App window should show the default screen with no camera selected.

**VIDEO**
https://user-images.githubusercontent.com/75075102/143402018-78469423-94fc-4beb-a9db-c1e90f35b867.mov

 - Version - CC version 1.3.1 [6], macOS Monterey 12.0.1
 - Camera - Fingers 1080 Hi-Res Webcam 

**Additional context**
I have only tried to test with 1 external webcam, I don't own any other external webcam

rcrodrix, 2021-12-21:
I was able to reproduce the issue, as written by bug originator. 
I was also able to find a temporary workaround.

Workaround:
With external camera unplugged:
Start application and select default system camera (in my case, FaceTime HD Camera)--verify this works.
Connect external camera (Kiyo Razer) and select it in dropdown.
Verify all app functions work with the external camera, as well as the default system camera.

Closing the app or disconnecting the external camera brings about the original app crash until the workaround is implemented.

CC version 1.3.1 (6), Mac OS Big Sur 11.6.1
Camera: Kiyo Razer

rpetty, 2022-02-07:
Followed the workaround instructions above (Mac OS 12.2). Verified FaceTime HD Cam is working.  Plug in Caldigit TS3+ with external camera, Logitech Brio 4k and app crashes.  Happy to provide additional debug information.

closed 2022-03-31 by stale[bot]

syzmekk, 2022-05-18:
Same here - MBP 2021 16" M1 Pro with the Desptech camera.