itaybre/CameraController #30: Feature request: UI should expose (editable) numbers

wookayin, 2021-11-25:
[image]

All the settings are configurable via sliders, which do not provide an exact number. The UI should expose readouts of numbers (and preferably their valid ranges if space allows) and they should be editable to allow users to enter the values manually. Of course, sliders and text boxes should be synchronized with some reasonable sanity check.

This is an important feature for a more fine-grained control. Thank you!

wookayin, 2022-01-06:
@Itaybre still yet?

Itaybre, 2022-01-19:
I just built a version with this, but I cannot test it because I don't have my webcam with me, do you mind giving it a test?

Double Tap the controller name to be able to edit it
[CameraController.zip](https://github.com/Itaybre/CameraController/files/7898073/CameraController.zip)

Itaybre, 2022-01-19:
@wookayin please use this one, that one didn't link the field with the camera 
[CameraController.zip](https://github.com/Itaybre/CameraController/files/7898946/CameraController.zip)
values

wookayin, 2022-01-20:
@Itaybre Are you sure this is the correct build? The UI hasn't changed.

wookayin, 2022-01-20:
Oh I see, when you double click it toggles between slider and textbox. Personally I don't feel this is an intuitive design. Why don't you put the input box next to the sliders so they can be shown at the same time? IMO there is no need to toggle...

Itaybre, 2022-01-20:
@wookayin I wouldn't consider it final, but I don't want to clutter the interface with too many buttons.

I am open to options, like enabling this from preferences

closed 2022-03-13 by stale[bot]

wookayin, 2022-03-13:
@Itaybre could you please this re-open unless you really don't want to not implement this? (Personally I don't like stalebot keep closing the issues..)

Eloren1, 2025-10-08:
??????????????? I literally cannot check my shutter speed. Who thought about doing it sliders only?

Eloren1, 2025-10-08:
@itaybre

Eloren1, 2025-10-08:
> don't want to clutter the interface

A good interface is simple to use, it should be comfortable for both pros and noobs, and outputting exact values is useful for everyone. Blank sliders are so weird.