itaybre/CameraController #31: Adding to Homebrew Cask apps collection

mr-eshua, 2021-12-13:
Hello @Itaybre!

I would want to sincirely thank you for your App! Basically it gave new life to my Foscam W25 UVC webcam in MacOS 🙏

In return, I would like to propose adding it in [Homebrew Cask list](https://formulae.brew.sh/cask/) to be able to install it through `brew install --cask` command, of course, if you do not mind. A Cask script for that is completely on me :wink:

Cheers!

Itaybre, 2021-12-14:
I am glad you found CameraController useful.

It would be nice to be capable of installing with homebrew

mr-eshua, 2021-12-16:
The latest version `1.3.1` was added to [Homebrew Cask Drivers tap](https://github.com/Homebrew/homebrew-cask-drivers).

To install with Homebrew:
1. Add Drivers Tap first: `brew tap homebrew/cask-drivers` (does nothing if the tap was already added previously)
2. Use `brew install --cask cameracontroller` to install the app

I also took some liberty to update README with Homebrew installation information https://github.com/Itaybre/CameraController/pull/32.

Itaybre, 2021-12-30:
Thanks for adding this 💪

closed 2021-12-30 by Itaybre