itaybre/CameraController #33: HDR control for Razer Kiyo Pro

juliussohn, 2022-01-05:
Hey unfortunately the Razer Synapse Software to configure the webcam only works on windows. Camera Controller already covers most controls, but one key feature of the Razer Kiyo Pro cam is HDR, which would be awesome to have in camera controller

Itaybre, 2022-01-10:
I would need to see the data sent to the device to see if I can replicate it.

Do you mind extracting it with this software: https://freeusbanalyzer.com

Itaybre, 2022-01-11:
I just found that HDR is part of the UVC protocol, I will take a look.
But since I cannot test it, I can't promise anything.
I will need the output from USB Prober, see this small guide: https://github.com/Itaybre/CameraController/issues/25#issuecomment-953162336

juliussohn, 2022-01-11:
Does this help?

[image]

Also here are the settings from the Windows Software
[image]

Itaybre, 2022-01-11:
> Does this help?
> 
> 
> 
> [image]
> 
> 
> 
> Also here are the settings from the Windows Software
> 
> [image]
> 
> 
> 
> 

Yes, that work, but can you expand the "Controls Supported" in the VDC Input Terminal section?

juliussohn, 2022-01-11:
[image]

Itaybre, 2022-01-19:
I may have found some insight about how to control HDR and also FOV.

I will need help testing that, I couldn't find the camera at a reasonable price locally

juliussohn, 2022-01-19:
Sure, happy to test if you have a build!

Itaybre, 2022-01-19:
@juliussohn You should see a new section inside Advanced, for nor I just implemented setting HDR, please tell me if it works correctly

[CameraControllerRazer.zip](https://github.com/Itaybre/CameraController/files/7899597/CameraControllerRazer.zip)

juliussohn, 2022-01-21:
I cant see an HDR setting. Am I missing something?
[image]

Itaybre, 2022-01-21:
@juliussohn Seems like I used an incorrect value for productId and vendorId (to recognize the camera).
Do you mind sharing the values from USB Prober?
They should be at the top when selecting the camera

juliussohn, 2022-01-21:
Still nothing showing up

Itaybre, 2022-01-22:
@juliussohn found the issue, it should appear now
[CameraController.zip](https://github.com/Itaybre/CameraController/files/7917382/CameraController.zip)

juliussohn, 2022-01-25:
@Itaybre  The toggle is there, but it doesn't seem to change the HDR setting

Corbot5000, 2022-01-27:
I can confirm that turning HDR on and off doesn't change anything. It's worth noting that the Kiyo Pro only supports HDR at 30 FPS - could this be why it isn't working correctly? Is there any way to manually adjust the cameras to 30 FPS within CameraConttoller?

Happy to do what I can to test this functionality.

Itaybre, 2022-01-27:
@Corbot5000 I didn't know about the 30FPS rule.
I would need recordings made with [Wireshark](https://www.wireshark.org), toggling the FPS up and down.
And probably more recordings of enabling and disabling HDR.

I don't have such camera to test it myself, and it isn't even sold locally

Corbot5000, 2022-01-27:
@Itaybre I have wireshark installed and the window's camera app lets me toggle between "hdr pro" and "hdr off" when I take photos. If you could give me a little more guidance on how to analyze the Kiyo pro in Wireshark, I'd be happy to do it. I can be reached on Discord at Corbot8000#8015

Corbot5000, 2022-01-27:
@Itaybre attached are two wireshark logs -
1) KiyoProHDRProOn.pcapng - HDR Pro already turned on, a few photos taken, video recorded
2) KiyoProHDROff.pcapng - Turned HDR Pro off, photos taken, video recorded, HDR Pro turned back on

https://drive.google.com/file/d/1b6Jehiyer1171aHhA1ePDN-09gIP-Iyk/view?usp=sharing

Itaybre, 2022-01-27:
@Corbot5000 can you do the recording as short as possible? Like just pressing the toggle in the app and wait a couple seconds doing nothing
seeing too many rows doesn't help me find the correct ones

Corbot5000, 2022-01-28:
@Itaybre Here's the smallest recording I could do of HDR Pro being turned on and off.

https://drive.google.com/file/d/1jLbzS2a_tifFFbSZsh2uj9qDH58_bwqf/view?usp=sharing

carlosnasillo, 2022-02-02:
Have a Razer Kiyo Pro myself (on Mac M1), and happy to help debug or provide test data as required.

lupus2k, 2022-02-17:
Razer Kiyo Pro and Mac M1 Max user here, let me know how can I help testing.

authorfunction, 2022-02-23:
Hi, ordered a Razer Kiyo Pro recently and will be happy to help with testing as soon as it arrives.

soyersoyer, 2022-03-29:
You can find some working code to set the Kiyo Pro custom controls: https://github.com/soyersoyer/kiyoproctrls

authorfunction, 2022-03-29:
I tried the code @soyersoyer provided, but there's no /dev/video0 or any other obvious video devices in the /dev directory. The camera is on and working (I tried with it off as well). I am on Monterey (Apple Silicon).

jaketame, 2022-05-13:
I have a Kiyo Pro now and would be happy to test

mattheworiordan, 2022-05-22:
I would be happy to test too 👍

mk3rnb8, 2023-01-04:
Hi All, is there any progress? I'd love to use HDR with my MBP.

cheers,
M.

chansearrington, 2023-02-24:
Joining in on this thread. I just got two Kiyo Pro Ultras and love them. But it's hard to adjust anything on my Mac. Thank god for this app and eCamm Live

retpolanne, 2023-06-09:
Hey @Itaybre, would you mind sharing the HDR work on a branch so we can help? As @soyersoyer mentioned, there's this repo with some stuff for HDR and FoV that we can try to implement here

https://github.com/soyersoyer/kiyoproctrls/blob/main/kiyoproctrls.py

Itaybre, 2023-06-14:
> Hey @Itaybre, would you mind sharing the HDR work on a branch so we can help? As @soyersoyer mentioned, there's this repo with some stuff for HDR and FoV that we can try to implement here
> 
> 
> 
> https://github.com/soyersoyer/kiyoproctrls/blob/main/kiyoproctrls.py

I changed computers several times since this, so I might have lost the code.
But seeing there is a project with this working, it should be much easier to implement. Will try making a build this week

alexbochel, 2023-09-05:
I would love to second adding this feature. I am a bit rusty with xcode development but I will see if I can find a solution using the linux versions of this app.

padurean, 2023-12-12:
Are there any news about this?