itaybre/CameraController #38: Settings are not persistent

stffffn, 2022-01-29:
**Describe the bug**
The applied camera profile settings are lost across different apps/websites. The settings are only persistent for one app/website and will be reset if you open another app/website.
Tested with Photo Booth, Discord and Google Meet.

Another question: Is CameraController meant to be run minimzed in the background? Otherwise I'm not able to achieve persistent settings.

**To Reproduce**
Steps to reproduce the behavior:
1. Open CameraController, apply your profile settings and minimize the app
2. Open an app or website which uses your webcam
3. The settings are now back to default
4. Reapply your profile settings
5. The settings are now persistent for this app/website if you close it and reopen it again
6. Close this app/website or deactivate your camera
7. Open another app/website
8. The settings are back to default again
9. Reapply your profile settings, close this app/website or deactivate your camera and open the first app/website again
10. The settings are back to default again

**Expected behavior**
The applied settings should be persistent across different apps/wesbites.

**Setup:**
- CameraController 1.3.1
- Logitech C920
- MacBook Air M1
- macOS 12.1
- Photo Booth 12.0
- Discord 0.0.264
- Chrome 97.0.4692.99 (used for Google Meet)

jonwest, 2022-03-01:
Same behaviour here

JensRoland, 2022-06-09:
Is there really not a way to configure this in the tool now? I use Google Meet for work and since every meeting is a new Chrome tab, the settings aren't persisted. Reapplying before every call I have is not sustainable for me, so persistent settings is really a must for me.

Then again, it's a free tool and it does a great job when it's active, and beggars can't be choosers.

benmosher, 2022-11-22:
FWIW, in my case I have a Razer Kiyo X and it seems to be that Google Meet (in Chrome or Safari) and Discord negotiate for the 720p 60fps mode, whereas Photo Booth and Zoom both seem to use the full HD 1080p mode. 

I did some digging and it appears that the AVFoundation classes allow applications to choose one of potentially many preset configurations and I suspect it's this action that dislodges any settings previously set in CameraController.

If I have time and energy, I may dig into this, but I am suspicious there's nothing that can be done in userland app code as this is likely a consequence of the way AVFoundation is implemented in macOS.

Anyway, well said @jamesgeddes, I am pleased as punch to have this app at all, as the feed looks terrible without it 😁. much appreciated, @Itaybre.

benmosher, 2022-11-22:
FWIW this SO answer captures most of what I have been able to dig up so far: https://stackoverflow.com/a/40097057/344143

If anything, it suggests that maybe running CameraController in the background with the configuration locked and no capture running might be an option. _maybe_. 🤞