itaybre/CameraController #42: Programm Crash when started

Scybe1, 2022-02-08:
Hey together- I am new here. i was searching for absolutely the programm you coded. Problem is, just the version 1.2 is working. higher ones does not work. I am Using MAC OS 10.15.7 on a Macbook Pro 2013 16 GB.

Is there something i could change to run the programm ?

Thank you very much !!

joekw, 2022-03-31:
I'm seeing this too. Looks like `uvcUnit` is set to -1.

atchinson, 2022-04-01:
Do you have OBS Studio installed on your Mac or some other virtual non usb camera? Sometimes the app works, most of the time it crashes complaining about a negative number.

My other Mac without OBS seems to work fine.

jazzychad, 2022-04-28:
I have opened a PR for this issue: https://github.com/Itaybre/CameraController/pull/57

jazzychad, 2022-05-05:
This can probably be closed now? (unless you want to cut another release first?)

Itaybre, 2022-05-05:
You are right, closing it now

closed 2022-05-05 by Itaybre