itaybre/CameraController #47: Not working with multiple camera of the same model

Zandalus, 2022-02-17:
I've attached multiple Logitech C920 and I can't change their parameters properly.

If only one of them is attached everything works properly, but as soon as I attach both it stops working (one of the two doesn't change his parameter anymore).

Itaybre, 2022-02-23:
Both cameras have the same model and name?
It is possible that the USB lookup is not working for the second camera in that case

closed 2022-04-16 by stale[bot]

michaelbaisch, 2022-07-06:
I would like to reopen this, since I have the same problem, with two identical Webcams.

I did a bit of digging and the problem seems to be the [getIOService()](https://github.com/Itaybre/CameraController/blob/master/CameraController/UVC/Extensions/AVCaptureDevice+USB.swift#L33) method, for both cameras the same number is returned. I looked into the properties from `IORegistryEntryCreateCFProperties` of both cameras, here is the diff: 

```
@@ -1,7 +1,7 @@
 ▿ 28 elements
   ▿ 0 : 2 elements
     - key : sessionID
-    - value : 5957302424021
+    - value : 5957309576006
   ▿ 1 : 2 elements
     - key : USBSpeed
     - value : 3
@@ -55,10 +55,10 @@
     - value : HD USB Camera
   ▿ 12 : 2 elements
     - key : USB Address
-    - value : 3
+    - value : 5
   ▿ 13 : 2 elements
     - key : locationID
-    - value : 1114112
+    - value : 1310720
   ▿ 14 : 2 elements
     - key : bDeviceSubClass
     - value : 2
@@ -102,7 +102,7 @@
     - value : IOCommand is not serializable
   ▿ 26 : 2 elements
     - key : kUSBAddress
-    - value : 3
+    - value : 5
   ▿ 27 : 2 elements
     - key : kUSBVendorString
     - value : HD USB Camera
```

Is there any chance one of those could be used in the lookup?

Greetings