itaybre/CameraController #50: Logitech Brio - Focus not working

ptoharia, 2022-03-08:
Please enter the following data:

Brand: Logitech
Camera: BRIO
Have you tried many USB ports? Yes

The camera works, but the image is unfocused and the focus control does not change anything.

Itaybre, 2022-05-07:
@ptoharia does the focus change when autofocus is enabled / disabled?

ptoharia, 2022-05-07:
No, it does not change.
Regards
Pablo

superevr, 2022-05-23:
autofocus works just fine with my brio, so the cause could be something else.

sobarber, 2024-01-28:
I have Sonoma and Brio using 1.4.0 release with no problems so far.

billgertz, 2024-09-18:
I've turned autofocus off and updated a profile. At start autofocus is always on. Tried loading the profile with autofocus set on and after loading the profile by hand it is never set to off. I'm guessing that the profile doesn't retain autofocus settings, or it is being ignored.