itaybre/CameraController #51: OBS Studio Virtual Camera

atchinson, 2022-03-25:
Please enter the following data:

Brand: OBS Studio Virtual Camera
Camera: 
Have you tried many USB ports? Virtual....

2022-03-25 18:08:38.833370-0500 CameraController[2004:54153] Error loading /Library/CoreMediaIO/Plug-Ins/DAL/obs-mac-virtualcam.plugin/Contents/MacOS/obs-mac-virtualcam:  dlopen(/Library/CoreMediaIO/Plug-Ins/DAL/obs-mac-virtualcam.plugin/Contents/MacOS/obs-mac-virtualcam

I don't need to control the virtual camera, was looking for away to skip it since it is virtual and not actually plugged in.
I have checked the System Report to see what it sees plugged in and the OBS Studio Virtual Camera is not listed.

UVCControl - line 41
wIndex: UInt16(uvcUnit<<8) | UInt16(uvcInterface,    --- Error --- Thread 1: Fatal error: Negative value is not representable

Itaybre, 2022-04-01:
This probably happens because the Virtual camera does not have the USB endpoints

closed 2022-04-01 by Itaybre

atchinson, 2022-04-01:
Decreasing what / which resolution? The Mac's desktop native resolution or the camera resolution?

atchinson, 2022-04-01:
Is there way to test for that negative number or a virtual camera and skip it from detection / loading / probing for it's settings?