itaybre/CameraController #52: C922 Pro - Zoom Issue?

atchinson, 2022-03-25:
Please enter the following data:

Brand: Logitech
Camera: C922
Have you tried many USB ports? Many

The camera works. All the setting work. The issue I possibly see is with the Zoom. I didn't seem to zoom as far as I had seen on a Windows PC. I put some extra GUI stuff on the Advanced tab to actually show the Zoom min and max, what the program goes and gets from the camera. I also put a 3rd text object to show the current Zoom number setting. 

min = 100
max = 500

However, zooming basically stops at 120. The slider will keep going to 500, but no more zoom.
Is Logitech just being weird? I also hard coded the slider values and allowed the slider to go lower than 100 just to test, it would not zoom out any further past 100.

Itaybre, 2022-04-01:
Sometimes Zoom depends on the video resolution, decreasing it allows to use more zoom.
I never added an option in the app because usually you will want to do that from the Main App you are using the camera

Can you confirm if decreasing the resolution you are able to have more zoom?

atchinson, 2022-04-01:
In my system preferences for display, I tried Default for Displayed and Scaled with different resolutions.

The same zoom issue occurs.

atchinson, 2022-04-01:
I install logitech's Ghub app. I allows you to zoom from 100% to 500%. When I added the text values to the app I could see min at 100 and max at 500. But the slider stops at about 120 before it stops zooming.