itaybre/CameraController #53: GoPro Hero 10 not recognized

FezVrasta, 2022-03-30:
Please enter the following data:

Brand: GoPro
Camera: Hero 10
Have you tried many USB ports? Yes

Itaybre, 2022-04-01:
CameraController requires the camera to comply to UVC Standard.
I don't really know if the GoPro meets the criteria, for example, can you use it for FaceTime / Photo Booth?

FezVrasta, 2022-04-01:
Thanks for the answer. No I can't used it with Photo Boot or FaceTime, I can use it with any browser, Zoom, and other apps though.

olebedev, 2023-01-05:
Same problem with GoPro 8 Hero Black.

closed 2023-01-05 by Itaybre

olebedev, 2023-01-05:
Any chance you can look at make the CameraController work with  not UVC compliant cameras like GoPro? That would be awesome.

Anyways, thanks for the great project!

Itaybre, 2023-01-05:
@olebedev Probably no, it requires the camera itself to have such feature, usually it is down with the UVC standard.

Browsing GoPro website, we can see that their utility doesn't even support changing these settings, so the answer is now (but if they add support via a firmware update in the future, I can add support for it)

olebedev, 2023-01-06:
@Itaybre, got it, thanks for the explanation. Good to know :)