itaybre/CameraController #64: Doesn't open on M1 Pro

Rawnly, 2022-05-23:
**Describe the bug**
Doesn't open. it "quits unexpectedly"

**To Reproduce**
Steps to reproduce the behavior:
1. Open the app 

[Here a crash log](https://pastebin.com/sPuv5KWg) (pastebin password is: `camera controller`)

**Expected behavior**
Open the app

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Version 1.3.1
 - Camera Aukey PC-LM1E

jazzychad, 2022-05-23:
This was fixed by edb2d4fdb3cb2547d2cfcfdc9d3b3d9dd4dc7fc2
https://github.com/Itaybre/CameraController/commit/edb2d4fdb3cb2547d2cfcfdc9d3b3d9dd4dc7fc2

It is available in the latest 1.4.0 release: https://github.com/Itaybre/CameraController/releases/tag/v1.4.0

Rawnly, 2022-05-23:
Ok thanks! You might need to update the homebrew release too (i installed it this morning and i got the 1.3.1!)

closed 2022-05-23 by Rawnly

jazzychad, 2022-05-23:
aha.. might need to file a separate issue for that!