itaybre/CameraController #66: None of the Controls work on M1 MacBook

homurahomu1, 2022-05-24:
Hey, I think this app is really nice, but I recently downloaded it on my M1 MacBook Pro(14") to tweak some of the camera settings and nothing works. The preview shows, but none of the controls work at all. If someone could fix this I would be very grateful
[image]
!

jazzychad, 2022-05-25:
AFAIK, this app does not work with the FaceTime camera, but only with other UVC-compatible USB webcams attached to the computer.

Itaybre, 2022-05-25:
It is as @jazzychad said, apple doesn't allow communication with the FaceTime Camera without special entitlements.
This is why CameraController cannot change settings for it

closed 2022-05-25 by Itaybre