itaybre/CameraController #69: Problems with manual exposure and gain for Logitech c920

onlyway, 2022-06-30:
**Describe the bug**
For Logitech c920 I have problem. Auto exposure settings are overexposing image, I'm trying to use manual one. But event setting gain and exposure to 0 makes screen almost white on camera. 

**Expected behavior**
Exposure and gain settings should give ability to configure image properly. 

**Desktop (please complete the following information):**
 - OSX 12.4
 - Camera Logitech C920

Zeitkind, 2022-07-14:
I have a similar problem with a Microsoft LifeCam Studio. Auto exposure gives an kinda usable picture, manual is either black (exposure 0) or overexposed as soon as I try to move the slider, it's like slider position left is 0%, position anywhere else is 150% and no change at all.
MacOS 10.15 (Catalina)

neilo-df, 2025-02-09:
I have this problem with a generic uvc webcam based on a Sonix chipset.  It looks like this application isn't properly handling some of the UVC controls.

I wrote a short python program to dump the camera controls, and this is what I see.

The Exposure Time control:
```
Absolute Exposure Time
	 value: 5000
	 min: 1
	 max: 500
	 step: 1
	 default: 157
The `time` parameter should be provided in units of 0.0001 seconds (e.g., use the value 100
for a 10ms exposure period). Auto exposure should be set to `manual` or `shutter_priority`
before attempting to change this setting.
```

There's a range of 1-500, but with the slider all the way to the right I'm getting a value of 5000.  Which means I have to restrict myself to adjustments only in the first 1/10 of the slider's range if I want a valid exposure.

Another case, here is Brightness on my camera:
```
Brightness
	 value: 32
	 min: -64
	 max: 64
	 step: 1
	 default: 0
This is used to specify the brightness.
```

This UVC control wants a range of -64 - 64, but the slider can only produce 0-64.