itaybre/CameraController #73: Question about intended use, "open at login" and "read"/"push" settings

k8seren, 2022-08-17:
Hello! Apologies for a newb question. This software is super helpful for using my Razer Kiyo cameras with macOS. 
Is this program meant to be open at all times, i.e. run at login and always on? 
What is the function of the Read / Push settings under Advanced, and is it recommended these are enabled, or no? 

I'm not familiar with what many of the options actually *do* but messing with the sliders is intuitive enough. These options are the only ones I can't quite sort out on my own.  

I'm really grateful for this app and appreciate some clarification. Thanks for the hard work!

TheBoroer, 2022-08-30:
+1

I'm wondering the same thing

jmhodges, 2022-10-22:
Yeah, is there some way to turn off it running at boot? It's not listed in my login items but still manages to get booted on login.

iconmaster, 2023-03-15:
[From what the dev says](https://github.com/Itaybre/CameraController/issues/9#issuecomment-855427941), it sounds like the read/push settings have to do with accommodating how the camera may be adjusting its own values while you're using the app.

I'm going to leave Read disabled and Push enabled and see what happens.