itaybre/CameraController #79: Feature suggestion

Alvazz, 2022-10-11:
Sir, Can you add a function to set the camera resolution? This function is very useful! Very good app, Thanks!

Alvazz, 2022-10-15:
Sir,
I don't understand technology, and I don't know if this feature suggestion I put forward is very difficult to implement with technology. If the workload is not large, or the development difficulty is not high, it is strongly recommended to implement this function as soon as possible, because your This App is the best I have found to use. Unfortunately, there is no function to set the camera resolution, which is the only shortcoming.

aioue, 2023-01-24:
@Alvazz you can sponsor the developer: https://ko-fi.com/itaybre

davies-w, 2024-04-28:
I just came here to file the same feature request (I just got a new webcam, and I can't figure out if it is actually 4K or not)