itaybre/CameraController #81: All Options Are Greyed Out

mrwizzer, 2022-10-11:
This looks promising, but all the options are greyed out for me under both the Basic and Advanced tabs.  (Brightness, Contrast, Zoom, Tilt, etc.)

I'm using a Razer Kiyo as well as the built-in cameras on a 2-year-old iMac Pro running Catalina and a brand new MacBook Pro running Monterey, same results on both.

Thanks for any tips on getting this up and running.

-Mike

Itaybre, 2022-10-11:
It is normal for built-in cameras to not work, since apple doesn't allow apps to communicate directly with them.

But the razer camera should work fine, this happens on both computers?

mrwizzer, 2022-10-11:
> It is normal for built-in cameras to not work, since apple doesn't allow apps to communicate directly with them.
> 
> But the razer camera should work fine, this happens on both computers?
> 
> 

I stand corrected, it apparently works fine on the Razer on the iMac Pro, but not on the new MacBook Pro.  Is it because the MacBook Pro has an M1 chip?

Am I out of luck on the MacBook Pro?

Thanks for the prompt response!

-Mike

Itaybre, 2022-10-11:
That's odd, it should work fine. 

> It is normal for built-in cameras to not work, since apple doesn't allow apps to communicate directly with them. But the razer camera should work fine, this happens on both computers?
> I stand [...]

DohMoose, 2022-10-18:
> It is normal for built-in cameras to not work, since apple doesn't allow apps to communicate directly with them. But the razer camera should work fine, this happens on both computers?
> I stand [...]

I'm on an M1 Mac with a Kiyo. I just restarted the app and everything was enabled!

strongface99, 2022-10-26:
Hi, I am having the greyed out issue, but only on Zoom/Pan/Tilt. This is really what I am hoping to have as my camera has a very wide FOV which doesn't look great. 

I have a Lenovo Essential FHD Webcam, connected to an M1 Mac Mini. 

Thanks

dezerving, 2022-11-04:
> Hi, I am having the greyed out issue, but only on Zoom/Pan/Tilt. This is really what I am hoping to have as my camera has a very wide FOV which doesn't look great.
> 
> I have a Lenovo Essential FHD Webcam, connected to an M1 Mac Mini.
> 
> Thanks

I was able to get this to work by doing the following (not sure which one triggered it to ungrey everything):
- Gave CameraController "Full Disk Access" permissions
- Went to Control Center and made sure portrait effect wasn't on for the camera
- Restarted the app multiple times

I'm on a M1 MacBook Pro.

joekw, 2022-11-07:
It's weird, I get all the expected controls for my Kiyo Pro while running the last built release, but when running directly from Xcode, all the controls are greyed out.

It looks like it's failing to create the USBDevice. In AVCaptureDevice.usbDevice(), the interfaceRef is always nil.

levyjonlevy, 2022-12-07:
Zoom/pan/tilt greyed out for me as well. Laptop is a Macbook Pro 13" 2020 model, running Monterey, webcam is the built-in webcam on a Lenovo ThinkVision T24v-20 monitor.

annikolov, 2022-12-23:
Zoom/pan/tilt greyed out for me as well. M1 Mac Mini on Ventura 13.0.1 with Vitade 682H Camera.

noellered, 2023-01-06:
On MacBook Pro (~2019) Intel Chip this happens to me too. I've noticed it more when I have the camera open on another app before opening CameraController (i.e. Zoom, Google Meet)

chansearrington, 2023-01-29:
Just installed the app and all the settings are greyed out for me as well for both of my Mevo Start cameras over UVC.

I'm also on an M1 Macbook Pro.

About to try setting the app to launch automatically at login and restarting my computer.

1951FDG, 2023-02-02:
Same here, this happened when upgrading to macOS Ventura, also auto focus doesn't seem to be working anymore,

Im using a Razer Kiyo X

chansearrington, 2023-02-03:
> 

A restart fixed it for me.

mitkel, 2023-02-08:
Had the same issue. Additionally, if I evener managed to "ungray" the controllers, the auto focus option still didn't work for me. Found the following threads: [thread1](https://www.reddit.com/r/razer/comments/glte0s/razer_kiyo_wont_focus/) [thread2](https://www.reddit.com/r/techsupport/comments/gpyc4l/razer_kiyo_webcam_auto_focus_issue_can_anyone_help/) helpful. 

TL;DR - unpluging and pluging in the USB cable did the trick (instead of restarting).

1951FDG, 2023-02-08:
I can confirm, unplugging and plugging in the USB cable works! Autofocus working again!

1951FDG, 2023-02-09:
Opened CameraController app again today, options not greyed out, closed the app and reopened the app, options greyed out, but settings were still applied to camera.

madsenjk, 2023-12-21:
> I was able to get this to work by doing the following (not sure which one triggered it to ungrey everything):
> 
> * Gave CameraController "Full Disk Access" permissions
> ...

Thank you for this suggestion. Your first suggestion alone solved the grayed out issue for me. For the first 5 or 6 times I opened the app on first install it had worked fine, allowing full control of my external camera (Dell Pro 2K). Out of nowhere the next 5 times I opened the app there was no way to get the bars recover from greyed out - with no ability to edit. After coming here, and following your suggestion to allow "Full Disk Access" to the application all appears to be available again. 

CameraController 1.4.0 (6) - homebrew install
Ventura 13.2.1
Dell Pro 2K WB5023

quicksite, 2024-11-06:
Me too...  But only now have I read dev's comment that

> It is normal for built-in cameras to not work, since apple doesn't allow apps to communicate directly with them.
https://github.com/Itaybre/CameraController/issues/81#issuecomment-1275368397

Too bad. I did not know this.

martonpaulo, 2024-12-12:
Same issue here. All settings are grayed out.

I’ve tried using both the built-in Mac webcam and iPhone mirroring, but neither worked.

I also granted full disk access to CameraController, restarted the app, iPhone, and Mac, but it didn’t resolve the issue.

Device Details:
MacBook Air M1
Sequoia 15.1.1

Itaybre, 2024-12-12:
> Same issue here. All settings are grayed out.
>
> I’ve tried using both the built-in Mac webcam and iPhone mirroring, but neither worked.
>
> I also granted full disk access to CameraController, restarted [...]

You cannot control this settings in either the built in webcam or iPhone mirroring.
CameraController only works with USB cameras

hjjl26, 2026-02-24:
So MacBook Pro with M1 Pro running Sonoma 14.7.7 everything greyed out with USB camera Yongnuo YN 433 get image but no control, screen shot attached. Any thoughts? tried most suggestions up above. Thanks!

[image]