itaybre/CameraController #83: Feature Suggestion : Show as Virtual Camera

proactivematter, 2022-11-14:
Hello,

I have a specific usecase for which I require to route a camera through a lightweight virtual camera so that it may output to another program. I wonder if this is something that would be up for consideration? If not, is there another app that would be suitable? Again all I need is to be able to select a camera source and have it be converted into a virtual camera.

Asides from this, though I have no use for it currently, I just wanted to say this is a very cool app!

Many thanks.

jturnbull, 2022-11-14:
You can use [OBS](https://obsproject.com) for that (and much more)

proactivematter, 2022-11-14:
Hah. Sorry, my bad, I should probably have included this detail - the program I will be routing the virtual camera to is OBS.

I'll just go ahead and explain the whole situation, for anyone who may be curious.

OBS 28 + MacOS Ventura allows for use of Continuity Camera (using iPhone as camera, streamed to the Mac). Issue is that when you add the iPhone as a Source in OBS, the iPhone becomes locked in this mode (you can't use it for other stuff while Continuity Camera is running). Also because of the way OBS works, turning off a source does not turn off the feed. Essencially, if you add a source (camera in this case) to a scene, it will always be running. As a workaround, I tried creating a separate scene collection with only this camera, but switching scene collections during a Stream / Recording is not supported and results in instability / choppiness.

Basically what I am looking to do is to add to OBS a 3rd party Virtual Camera that takes the signal from Continuity Camera, and then only running that 3rd party app when I want to use the iPhone for this purpose. This I am pretty confident would solve this issue.

Itaybre, 2023-01-05:
As said by @jturnbull, you can use an OBS.

This feature is not relevant to what this project aims, creating a virtual camera is different from allowing users to control already existing settings in cameras

closed 2023-01-05 by Itaybre