itaybre/CameraController #84: App hangs with 100% cpu usage with my digital microscope webcam

emilycestmoi, 2022-11-24:
Version: 1.4.0
MacOS: 11.7

Hi! Just gave this a try and it starts fine without my digital microscope webcam connected (aliexpress generic). As soon as I connect my microscope the app will hang with 100% cpu usage when it's already started OR if I start the app after connecting the camera. The only way to get the app to run is if I disconnect my microscope webcam.

I noticed elsewhere you requested someone with this issue use USB.Prober.zip, I've done so and I've attached the output here "GENERAL - UVC" by "Genesys Logic, Inc".

[USB Bus Probe.txt](https://github.com/Itaybre/CameraController/files/10087670/USB.Bus.Probe.txt)

Thank you!

emilycestmoi, 2023-01-06:
I am a software developer so I may be able to debug, although I have no experience with macos dev. Any pointers as to what to look for? Thanks!

Itaybre, 2023-01-06:
I don't have any guess for this, but if you can pause the app mid running and see the running thread, that could give an insight about where the problem is.

You can pause if with this button on the bottom bar:
[image]

And then you will see the threads and method in execution on the left (in this case since the app is in standby there is nothing here, but with the CPU at 100% there should be something running):
[image]