itaybre/CameraController #85: Zoom Pan Tilt not working for Microsoft Modern WebCam

sadiqkhoja, 2022-11-27:
I am trying to set Zoom, Pan and Tilt for "Microsoft Modern Webcam" on MacBook Air M2, I can move the slider but nothing happens, I am attaching UBS Bus Probe here:

[USB Bus Probe.txt](https://github.com/Itaybre/CameraController/files/10099378/USB.Bus.Probe.txt)

[image]

Itaybre, 2023-01-05:
With the resolution at max, changing the zoom value does nothing.
This is how most cameras work, the digital zoom won't work if you are using the max resolution

closed 2023-01-05 by Itaybre

Itaybre, 2023-01-20:
@konopielko  You will need to do that on the app that is streaming, they usually have a configuration for that

Itaybre, 2023-01-20:
> > @konopielko You will need to do that on the app that is streaming, they usually have a configuration for that
> 
> I only use my webcam for video meetings, and zoom or google meet don't have settings like that - so is the only solution to use a program like obs as a pass though for the video feed?

Both Zoom and Google meet have an option for that