itaybre/CameraController #103: Unable to reset, turn off, or uninstall

otherlander, 2023-05-24:
CameraController settings stay with the camera after the app is deleted and when using the camera with other computers. Please provide a way to uninstall or guidance to remove files to reset camera. 

iMac with Ventura using iContact Camera

Itaybre, 2023-05-26:
You can reset the settings selecting the Camera Default profile and pressing apply:
[image]

otherlander, 2023-05-31:
Thanks very much for your help. I thought you might be interested to know that CameraController appears to be a product on reincubate's website https://reincubate.com/camo/