itaybre/CameraController #119: Development setup questions

Sab44, 2023-11-24:
Not a bug.

Found this software, have been using it regularly and wanted to contribute, as I'm a Swift dev myself.
Forked the repo, fetched the master branch and started getting to work.

1) current state of the master branch does not launch an application window
2) I have seen that there have been many changes since the last released version so I checked out v1.4.0 which allows me to launch the window. But all settings in the window are greyed out. After debugging I came to the conclusion that `CaptureDevice` initialization fails as `try? UVCDevice(device: avDevice)` is throwing an error for the usbDevice. Everything is working fine with the version installed via Homebrew with my Logitech C922.

Some help is appreciated. If I can get it to work I can also supply an update to the README with more instructions on the dev setup.

Thanks!

Techlorine, 2023-12-13:
Same problem here while buiding `v1.4.0`, adding some information  from the best of my knowledge:

[Here](https://github.com/Itaybre/CameraController/blob/v1.4.0/CameraController/UVC/Extensions/UnsafeMutablePointer%2Binterface.swift#L36C45-L36C45) the `IOIteratorNext` in `UnsafeMutablePointer+interface.swift` always returns 0, implying no valid iterator was found according to the [doc](https://developer.apple.com/documentation/iokit/1514741-ioiteratornext). This is the deeper cause of the nil `interfaceRef` and the `Domain=usbDevice` error.

I'm new to Swift and I would appreciate any help on this problem.

### Building Environment:

- Mac Mini with Apple M2 Processor
- macOS Sonoma v14.1.2
- Xcode Version 15.0.1 (15A507)

belak, 2024-01-16:
I've been trying to get this working as well, though I'm not a Swift dev by trade.

I ended up trying to disable code signing in the project (and all sub-projects) because I just want to run it locally (and I don't have a developer license) and it seems to compile fine, but when I run it, I get this error:

```
dyld[8588]: Library not loaded: @rpath/Sparkle.framework/Versions/B/Sparkle
  Referenced from: <6300585A-D131-3F9E-BAFF-1B4763FC34A3> /Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/CameraController.app/Contents/MacOS/CameraController
  Reason: tried: '/Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/Sparkle.framework/Versions/B/Sparkle' (code signature in <08843E26-C65A-3889-B17E-AC620C1B375C> '/Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/Sparkle.framework/Versions/B/Sparkle' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/PackageFrameworks/Sparkle.framework/Versions/B/Sparkle' (no such file), '/Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/CameraController.app/Contents/Frameworks/Sparkle.framework/Versions/B/Sparkle' (code signature in <08843E26-C65A-3889-B17E-AC620C1B375C> '/Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/CameraController.app/Contents/Frameworks/Sparkle.framework/Versions/B/Sparkle' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/CameraController.app/Contents/Frameworks/Sparkle.framework/Versions/B/Sparkle' (code signature in <08843E26-C65A-3889-B17E-AC620C1B375C> '/Users/belak/Library/Developer/Xcode/DerivedData/CameraController-cbbodqtrarjkjsdpfsnaosqntqsb/Build/Products/Debug/CameraController.app/Contents/Frameworks/Sparkle.framework/Versions/B/Sparkle' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs)
```

There are also a few swiftlint errors, but they can be worked around.

einarmagnus, 2025-03-05:
Just wanted to add that I am having the same issue.

Works with homebrew, but building it myself it shows the video feed of my cameras, but all the settings are greyed out.
Error seems to be the same as Sab44 describes.