itaybre/CameraController #138: Razer Kiyo Device Support

rtfmoz2, 2024-12-12:
Can you add support for this camera. It displays picture but all the control are greyed out.

Brand:  Razer
Camera:  Kiyo 
Connection: cable with USB-A
Device VendorID/ProductID:   0x1532/0x0E03
Device Version Number:   0x7913
Have you tried many USB ports? Yes
Platform: Apple M4 Mini Pro
Ports: Thunderbolt 5/USB-C rear port
Adapter: male USB-C to female USB-A short cable
Mac usbdiagnose below
[usbdiagnose.txt](https://github.com/user-attachments/files/18104245/usbdiagnose.txt)

wireflo, 2024-12-25:
Funny that this is the top issue - me being Kiyo owner myself in search of a tool that works ;-)
If you have any pointers for me how to approach this I would be happy to try my luck and work on a pr - but struggling a bit where to start in the codebase (last time i worked on macos applications was pre-swift times ;-)

samithoughtbot, 2025-05-18:
I purchased [Razer Kiyo](https://www.amazon.co.uk/dp/B076QK6489?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1) recently and it is working with the software on my macbook pro M2 chip.